sethvoltz/OctoPrint-DisplayPanel #1: Official Plugin Registration

jneilliii, 2020-12-04:
Great plugin, I saw your project linked in the OctoPrint Discord and think it would be super cool to get this plugin registered in the official plugin repository. I'm one of the maintainers and can assist with the process, but it's detailed fairly well [here](https://plugins.octoprint.org/help/registering/).

sethvoltz, 2020-12-04:
Sounds great, I’d love to get registered. I think I should write up the rest of the hardware side (wiring diagram and some other details so folks can make their own) then I’ll register from that footing.

jneilliii, 2020-12-04:
Yeah, having all that documented in the readme with pics/screenshots/diagrams is a good idea prior to official adoption.

sethvoltz, 2020-12-17:
I finally had a moment to write everything up, update the repo with wiring instructions, etc. Submitted here: https://github.com/OctoPrint/plugins.octoprint.org/pull/714

RejectH0, 2020-12-18:
Would it be possible to get the STL files for the case as part of the project?

jneilliii, 2020-12-18:
@RejectH0 probably should open a separate issue for that. I'm closing this now that the plugin is officially regsitered.

closed 2020-12-18 by jneilliii

sethvoltz, 2020-12-18:
Following up for @RejectH0 — the STLs as well as Fusion360 and STEP are available on my Thingiverse thing, linked in the Hardware section of the reader. Please open a new issue if you have problems with that and I’ll consider keeping versioned copies on here