sethvoltz/OctoPrint-DisplayPanel #2: Display not functioning...

Andy-ABTec, 2020-12-18:
Not sure if I've jumped too soon but I happened to have a couple of the displays to hand...

Installing the plugin doesn't appear to work in that once I've installed it the system requests a restart and then gives the impression that the plugin isn't installed.

Tried with the Plugin Manager, installing from URL and Zip - all give the same result.

Only mention I can find in the logs is: -
`2020-12-18 17:06:43,843 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/sethvoltz/OctoPrint-DisplayPanel/archive/main.zip `

Thoughts?

Andy-ABTec, 2020-12-18:
Sorry - me again... Forgot to mention - I was initially confused by your circuit but then I twigged that you'd put pin 1 bottom right rather than top left. Viz: https://www.raspberrypi.org/documentation/usage/gpio/

Might be better drawn like this:
[image]
or at least with pin 1 designated.

Just a thought...

sethvoltz, 2020-12-19:
😅 yeah, that’s a great idea. I built it off my notebook drawing which is laid out in the orientation of my physical position on my printer. It totally makes sense to both orient it the way you suggest and also I should add something that denotes where the Pi ports are just in case this is someone’s first Pi project.  

Thanks for the suggestion!

To your original post, have you enabled i2c in the raspi-setup command?

Andy-ABTec, 2020-12-19:
Glad to be of help...

As to the i2C issue, I already thought of that... it's enabled using raspi-config, and double-checked from the content of /boot/config.txt: -
```
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=off
dtparam=spi=off
```
Any other ideas? 

In the meantime, I'll find another pi and install a bare OctoPrint to make sure nothing is clashing and causing problems.

Andy-ABTec, 2020-12-19:
### WooHoo! it's working...

After a fair bit of messing about I bit the bullet and installed "i2c-tools" to use "i2cdetect" try and figure out what the Pi was up to... Turns out it was denying the existence of the display module completely. 

**I HATE CHINEESE PATCH LEADS!** - I know, **Rookie Mistake**, I should have known better, still, it's working now! 

So, on to the serious stuff, knowing almost nothing about these things as I do, is it possible to invert the display? (to put the connectors at the bottom) and to change the GPIO pins, I know, "he's just got it working and already he wants to change it". 

There is a method in my madness, I'm looking at a smaller mount in a different place. I have my fingers crossed that there will eventually be a web GUI for the plugin allowing changes like this as well as language etc...

Please keep up the good work - and as I said before, if there's anything I can do to help or test please holler!

Stay safe and have a good Christmas...

Andy B.

kaju666, 2020-12-19:
So how you solve it?

Install is succesful but after rebot plugin is not installed.

kaju666, 2020-12-19:
> So how you solve it?
> 
> Install is succesful but after rebot plugin is not installed.

OK solve it 👍 
1. I enabled I2C in raspi config
2. Installed i2c-tools
3. Checked with command sudo i2cdetect -y 1 that my i2c screen is wrong connected, Fixed it.
4. Needed to reinstall plugin
5. The only bug is that after install this plugin it is not displayed as installed.

a1co, 2020-12-19:
Same Problem here. 

I am totaly new to Raspberry PI GPIO Usage and haven't soldered for 20 years.

I have no idea what i2detect tells me:

root@octopi4:~#  sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

What could I do?

Andy-ABTec, 2020-12-19:
I think the "3c" is the address of the i2c device so it's being detected (mine was originally all "--" until I found my duff patch cord)

@kaju666 indicated that the software didn't seem to install without the address being detected, so I would try installing again and then reboot...

Other than that I'm as much in the dark as you are - sorry!

a1co, 2020-12-19:
Thank you!

now it works: Shutdown, unpluged all, check wiring, reconnected and booted. Now it works, but I do not know why now ...
i2cdetect dit not change.
It was not the first rebooted, but I did not (yet) reinstall
Maybe a plug did not connect?

Andy-ABTec, 2020-12-19:
Been there - Chinese patch cords, wouldn't mind betting that was the problem - glad it's working now...

Andy-ABTec, 2020-12-19:
Just found this on adafruit.com...

> The OLED driver circuit needs a small amount of time to be ready after initial power. If your code tries to write to the display too soon, it may not be ready. It will work on reset since that typically does not cycle power. If you are having this issue, try adding a small amount of delay before trying to write to the OLED.

Not sure if its related, but that's probably one for @sethvoltz ...

schranzfreak, 2020-12-19:
had the same issue that the display did not show anything.
but after enabling i2c in raspi config and a reboot it worked just fine!
the guide should be extended with these instructions.

Andy-ABTec, 2020-12-19:
I believe @sethvoltz already has this on his to-do list along with updating the drawing.

It always amazes me that what's obvious to some isn't to others, I nearly miswired this when I was first testing, but I guess any new project, especially one which includes hardware as well as code will teeth...

sethvoltz, 2020-12-20:
Thanks for the feedback everyone, I'm making updates to the graphics and I will update the installation instructions here and make sure they're updated in the plugin listing as well. I'll close this out once those are pushed up.

sethvoltz, 2020-12-20:
@Andy-ABTec By invert do you mean white background with black text or rotating it 180º? Both are possible but it wasn't designed for either. I hadn't planned on adding a configuration panel, however that would be a good reason to have one.

sethvoltz, 2020-12-20:
Diagram and instructions updated. Closing this out. PRs welcome to improve language

closed 2020-12-20 by sethvoltz