sethvoltz/OctoPrint-DisplayPanel #3: Installation failure

Hairyloon, 2020-12-19:
Plugin manager reports:

> ERROR: Could not find a version that satisfies the requirement Adafruit-Blinka (from adafruit-circuitpython-ssd1306->OctoPrint-DisplayPanel==1.0.0) (from versions: none)
> ERROR: No matching distribution found for Adafruit-Blinka (from adafruit-circuitpython-ssd1306->OctoPrint-DisplayPanel==1.0.0)
> Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

The log says much the same.

jghake, 2020-12-19:
I am also receiving this error during install so I'm tagging myself here. I'm very excited to get this working, such a cool project. I've been looking for something that does exactly this for the very same reasons!

Hairyloon, 2020-12-19:
I suppose it gives us a chance to assemble the hardware. 😉

sethvoltz, 2020-12-20:
Just to eliminate this first, are you running OctoPrint on Python 3? There is a link in the Readme with official instructions to upgrade your OctoPi install. Once you have confirmed Python 3, please try to reinstall and update here?

Hairyloon, 2020-12-20:
> Just to eliminate this first, are you running OctoPrint on Python 3? 

Well I thought I was... but apparently not. Or in any case, the upgrade process fixed the problem.  Thank you.

I'll leave this open for jghake for now.

jghake, 2020-12-20:
I'm in the middle of a print, I'll try shortly.

jghake, 2020-12-21:
Install worked here. Thanks!

sethvoltz, 2020-12-21:
Thanks for checking back in ☺️ closing

closed 2020-12-21 by sethvoltz