sethvoltz/OctoPrint-DisplayPanel #4: Feature Request - Configuration Panel

Andy-ABTec, 2020-12-20:
HI @sethvoltz, thanks for the plugin - I'm a really happy bunny now it's up and running...

To continue the discussion from [here](https://github.com/sethvoltz/OctoPrint-DisplayPanel/issues/2#issuecomment-748482442) - apologies, my bad etiquette, I should have started another issue at the time... 

A configuration panel would be really useful for a couple of reasons (that I can think of at least!): -

Firstly, it would be really useful to be able to rotate the display 180º - I have an idea for a different case design to suit my particular needs and if I could use the display upside down it would make more room for the buttons!

Secondly, it could be useful to be able to change the GPIO pins used by the buttons. (I realise that the I2C pins can't be moved!) I'm already using one GPIO pin and 5V power to control a relay module which turns the actual printer on and off and another for an emergency stop button. I know there are other plugins that use GPIO out there and it would seem to make sense to be able to choose which pin does what to prevent possible clashes and possibly make GPIO usage a little easier in terms of cabling.

Just a thought!

Thanks again for adding some great functionality to OctoPrint!

Stay safe, and have a good Christmas...

Andy B.

sethvoltz, 2020-12-20:
I probably won't be able to work on this over Christmas as I'll be stepping away from the printer for some R&R but if you get a chance, PRs are always welcome ☺️  -- Leaving this open as a feature request

Andy-ABTec, 2020-12-20:
I'll take a look, but don't hold your breath! Pre-retirement I was a Linux and Network Admin for many years, other than a bit of shell scripting, javascript, (n)awk and expect I have virtually no coding experience. Adding a configuration page to Octoprint leaves me cold, let alone linking it to actions!

Still, I'll see what documentation I can find, It'll keep me distracted over Christmas...

unclej84, 2020-12-31:
Hello,

I would also be interested in having some option to select other pins. Is anybody already preparing a PR for this? Otherwise I would have a look and try to create a basic panel for selecting non I2C pins and display address. If someone is working on it I could support.

Regards Jörg

Andy-ABTec, 2020-12-31:
Well, as expected it kept me distracted over Christmas, and I still have the headache to prove it - the "Plugin Tutorial" leaves me cold, I won't be playing anytime soon as much as I'd like to...

unclej84, 2021-01-01:
@Andy-ABTec are you able to replace the __init__.py file and add a template folder for testing? I prepared the settings gui and tested it on my installation but it would be better if someone else could also check before creating the PR.

unclej84, 2021-01-01:
One question regarding initial setup. Currently the pins are set by default. Would you prefer to have this defaults or all buttons deactivated on installation. I also use GPIO Shutdown and Filament Sensor plugins and the display panel plugin is loaded before. So the pins used by the other plugins are already in use when loaded and these plugins don't work. After changing the pins for display panel the other plugins don't get reloaded automatically so you have to restart octoprint for them to activate their pins. If disabled by default instead of default pins all existing plugins would start as before and the user could select free pins. After saving the button would be activated directly using the selected pins. So, what is your opinion on this topic? Thanks in advance for your feedback.

Andy-ABTec, 2021-01-01:
More than happy to test for you, dumb question -where can I find the files? - Sorry, I'm a hardware and networking guy... Still trying to find my way around Github!

unclej84, 2021-01-01:
I just commited them - so you should see them now

bluespaniel, 2021-01-01:
I'm in the middle of a print right now but once finished I will give it a go.

Andy-ABTec, 2021-01-01:
So far so good - the panel loads and looks ok (though I think you're right disabled by default is probably the way to go), just digging around for some push buttons...

bluespaniel, 2021-01-01:
Yes very well done, disabled upon installation would be a good idea so it can be configured to work around other projects.

Andy-ABTec, 2021-01-01:
Buttons work too, though I'm going to investigate the Pi GPIO a little more, at the moment we're hanging "aerials" on the pins, a simple resistor may help in a noisy environment like the guts of a printer.

unclej84, 2021-01-01:
Okay, then I will put disabled for the buttons as default. What do you think about the display address? I think the default could be 0x3c. I made it so that if the address is not available/working the display is not doing anything without breaking other stuff. If somebody changed the address or has another default it can be changed in the settings ui. I hope I can manage to get the dropdown of available addresses working until tomorrow because it would prevent misconfiguration. So if I do not get any negative feedback from your side until tomorrow I will open the PR for this.

sethvoltz, 2021-01-01:
Thanks everyone for working on this. I’m looking forward to the PR

Andy-ABTec, 2021-01-01:
It's definitely a huge improvement, now if some kind soul can add an option to rotate the display 180 Deg I'll be as happy as a dog with two tails (still looking at GPIO input filter circuits)

unclej84, 2021-01-01:
Simple 180 degree rotation? I think that should be easy to integrate directly but I will check. There is a command for image rotation in PILlow which could be called before screen updates depending on a checkbox value of the settings page.

unclej84, 2021-01-01:
@Andy-ABTec I integrated a rotation function - if you want you can grab the latest files, test and give some feedback on that.

unclej84, 2021-01-01:
Okay, I am done so far - there is now a drop-down list for the display address instead of the input field. I defined the two most common addresses I found - 0x3c and 0x3d. The files are commited and can be checked/tested. Below a screenshot. I will do the PR tomorrow.

[image]

Andy-ABTec, 2021-01-02:
Wow - Love it...  Now I can go back to Fusion 360 and finish my case design, with the display rotated (pins at the bottom) it leaves much more room for the buttons.

[image].

I think there's some improvement to be made with some resistors and a cap filtering the buttons, but I'll fiddle when I'm a little more awake.

I may also play with the language on the configuration page a little, I'm sure I can improve the readability but don't wish to step on anyone's toes being as I haven't done any of the heavy lifting...

(Not wishing to date myself, but come back Fortran 77 - OOP leaves me cold!)

unclej84, 2021-01-02:
As you are the native speaker, if you have proposals for improvements of the settings labels, just put them here and I will directly integrate them

sethvoltz, 2021-01-02:
@unclej84 this looks fantastic. Open a PR whenever you're ready and I'll start reviewing this weekend.

My recommendations for the English labels would be to swap the words for the four buttons, e.g. "Mode Pin", "Cancel Pin", etc. I would also make everything "Sentence case" instead of "Word Case" to make them consistent across all labels.

sethvoltz, 2021-01-02:
@Andy-ABTec that's a cool looking mini-case. I dig it. When you're ready to publish it to Thingiverse, be sure to link it to mine and I'll do the same back so folks can find your alternative easier. Once it's up, I'll also link it out from the Readme here.

commit 391ed2ce (2021-01-02): Merge pull request #12 from unclej84/main

[Feature] Configuration Panel

Closes #4
Closes #12
closed 2021-01-02 by sethvoltz