sethvoltz/OctoPrint-DisplayPanel #5: [Feature] Screen timeot

kaju666, 2020-12-20:
Hi

Can you add screen timeout?
My RPI is all the time on becouse i use enclosure plugin to turn off the printer.
So this oled screen is also all the time ON. #

sethvoltz, 2020-12-20:
This would probably make more sense as a configuration option as I prefer it to be on. However with #4 it seems some sort of preferences is in order. Added to the list.

sethvoltz, 2020-12-20:
Oh, and if you'd like to take a crack at it, PRs are always welcome ☺️

spiff72, 2020-12-21:
Just wanted to chime in and say a timeout for the screen would probably be a very good idea.  I don't plan to build this one, but it looks like a nice simple option for displaying info.  (I use [Octodash ](https://github.com/UnchartedBull/OctoDash)with a 7" capactive display).  I have used some of these OLEDs for little projects in the past, and they are notorious for burn-in.  I had a clock running on one for about a year (maybe two), and the display was basically shot after that.  It isn't really appropriate for this project, but I have been using TTGO T-display [modules ](https://github.com/Xinyuan-LilyGO/TTGO-T-Display)(an ESP32 with a small TFT display).  The display uses an SPI bus connection so that may make the wiring more complex for a Pi.  

I used them for several [WLED projects](https://github.com/Aircoookie/WLED/blob/master/usermods/TTGO-T-Display/README.md) to display the IP address, presets, SSID, and color palette, but I also incorporated a timeout so the display disappears after about 5 minutes out of my fear of burn-in.

unclej84, 2021-01-02:
What is the idea at the moment. To generally disable the display after a certain (configurable) amount of time and enable on button press? Or deactivate after print finishes and activate on print start?
@sethvoltz as you have it on the list, do you already some kind of idea about the "specifications"?

spiff72, 2021-01-02:
> What is the idea at the moment. To generally disable the display after a certain (configurable) amount of time and enable on button press? Or deactivate after print finishes and activate on print [...]

I can't speak to what the others had in mind, but I think a configurable screen timeout delay would make sense.  For my project with the TFT screen and timeout I just hard-coded a 5 minute delay.  I did NOT take the time to configure a wake up function that would turn it on, other than just pressing a button.  (That was beyond my scope of knowledge of the entire WLED project and would have required coding beyond the usermod file that I was creating).  If there was a way to code things such that a button press WHILE the screen was off would ONLY wake the screen and then subsequent presses would operate as the buttons would normally operate, this would be ideal in my opinion.  For my WLED application, you basically have to press the button and know that it will do its normal function (either turning off the LEDS or cycling to the next preset) to get that screen to wake up.

For this printer status project, I think that leaving it on whenever the printer is operation could be an option, but would still risk burn-in of the OLED over time - especially if the printer is operating a lot, or runs a lot of long print jobs.

sethvoltz, 2021-01-02:
For me, personally, I would like the display always-on during a print (which is why I built it) but I think a 5min or shorter timeout would be perfectly reasonable when not printing. Spitballing, I would suggest two settings: 

* Timeout in minutes
* Dropdown for timeout on printer disconnected, connected but idle, or always

spiff72, 2021-01-02:
Just to add to my comment above, it looks like the "M" button changes the display mode.  That would actually be a perfect candidate for the wake-up button - even if it would cycle to another screen at the same time it woke up.

sethvoltz, 2021-01-02:
Reading @spiff72's description, I would add that there are two ways to consider a wake: any button would wake but not perform its action until pressed again, and only wake on Mode press, any other button press would be ignored while idle. I personally think the first option is better so if I implemented it, I would go that way.

sethvoltz, 2021-01-02:
Yep, the M (Mode) button is just a display button. I think we're on the same page for that, but I would expand it as in my comment above to also allow any button to "wake" the screen, then you would press them again to perform the desired action. 

@spiff72 what do you think of the two options suggested above?

unclej84, 2021-01-02:
If nobody is on it I would try to start working on it this afternoon after the cleanup for #4 is done. Dropdown and time select will not be an issue to implement. A wakeup on any button will also be quite easy (just add a call to the wakeup method and then quit the normal method of the button, if sleep was active). The trigger for the timeout should also be no big deal as there is already some timer for the screen update which can be used as template.

spiff72, 2021-01-02:
I think any button to wake would probably be preferred with your caveat mentioned above (button wouldn't perform its intended function with that first press).  My only concern would be an accidental double-press of a button like the cancel button (but that one appears to be coded to require a confirmation press too).  So an accidental triple-press of the cancel could cancel a print inadvertently.  Probably not very likely - but if the buttons are not debounced in some way, this technically could happen?

unclej84, 2021-01-02:
We could add a small sleep (e.g. 0.5s) to the wake-up method before releasing the marker. So even if the button triggers twice within this period of time it would not pass to the real method before the marker is finally removed.

sethvoltz, 2021-01-02:
The debounce is built into the RPi GPIO library, so that's already taken care of for us. I also handle all button presses in the same method then call out based on which pin was pressed, so wake would be very easy to trigger.

As for how sleep/wake would work, on sleep it would just set a bool and blank the screen (there's no powering it down or disabling it like a LCD) and on wake, toggle the var again and call the render function.

unclej84, 2021-01-02:
The ardafruit library has a [power off method](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/a92acaf468f3cb2d3c5de0adcb5ccf4d230f6cd0/adafruit_ssd1306.py#L143) for the diplay so it looks like it can be turned off.

hallskelet, 2021-01-03:
I think the screen resource will be enough for the time while printing is in progress.
I am using the "PSU Control" plugin to shutdown the printer after printing and it changes the state of the system when shutdown.
It might be useful to use the printer state (connected or not) to turn off the display when the printer is automatically turned off after printing.

unclej84, 2021-01-03:
I am currently testing the code so the files should be available for testing within the next hour.

unclej84, 2021-01-03:
Code is commited in my [Repo](https://github.com/unclej84/OctoPrint-DisplayPanel/tree/feature-timeout) and can be tested. The tabs of #13 are already included.

sethvoltz, 2021-01-16:
Merged

closed 2021-01-16 by sethvoltz