sethvoltz/OctoPrint-DisplayPanel #6: Enhancement - OLED Display models

Andy-ABTec, 2020-12-21:
Hi @sethvoltz - apologies back again...

Whilst I was really impressed with your case design, unfortunately, it doesn't fit my needs and I started rummaging around on the internet looking for drawings for the display.

Eventually, I found one and from that, I created basic Fusion and STEP models which I've attached here. Please feel free to include them in the project if you wish.

With regard to a PR for a configuration page, I'll have a go but at the moment I'm working my way through @foosel's tutorial in the vain hope of understanding what's going on!
[image]
### [Models.zip - the only way I could see of getting the Fusion and step files here](https://github.com/sethvoltz/OctoPrint-DisplayPanel/files/5724980/Models.zip)

bluespaniel, 2020-12-22:
Thank you, that is very useful.

Andy-ABTec, 2020-12-22:
PLEASE DELETE THIS _ THERE'S OBVIOUSLY MORE THAN ONE SIZE OF 0.96" BOARD - Taking of boards, back to the drawing board, well Fusion anyway...

Andy-ABTec, 2020-12-22:
PLEASE IGNORE THIS ISSUE - DRAWING OF SLIGHTLY DIFFERENT SIZED BOARD...

closed 2020-12-22 by Andy-ABTec