sethvoltz/OctoPrint-DisplayPanel #7: Possible debounce problem

towlerg, 2020-12-22:
Looks like a nice project particularly for situation where the printer is far from PC. One small thing, I wonder if hitting cancel twice is optimal? Would it not be safer to have to hold the cancel button down for some number of seconds.

sethvoltz, 2020-12-24:
Yep, I found the software debounce in the underlying library to be... finicky. Not much I can do there so I set it to 250ms to avoid it double clicking on a single press. If you double-click the cancel button it will likely ignore the second one and you'll have to press again within 10 seconds.

If you are still having problems with a long wire, try throwing a capacitor across the wires to help hardware debounce.

sethvoltz, 2020-12-29:
@towlerg checking in to see if you are still having debounce issues?

towlerg, 2020-12-29:
Hi Seth,  I'm sorry I appear to have mislead you. I have not implemented you plugin, I was expressing a concern that in principle usinf a double click to cancel was perhaps not optimal. Personally I'd prefer a very long press (like 10 secs). It changes the most obvious fault mode from debounce (all switches bounce) appearing as a double click to stuck switch.

By all means close this if you want but perhaps if you're happy with a double click you could add long press as an option?

sethvoltz, 2020-12-29:
I'm going to close this for now but I'll consider it after I get to some of the other feature requests and there is a preference panel in place.

closed 2020-12-29 by sethvoltz