sethvoltz/OctoPrint-DisplayPanel #8: Screen going blank and random pauses during prints

bluespaniel, 2020-12-24:
I am experiencing some issues with the plugin. The screen goes blank at random times during a print and the prints also pause at random intervals. If I reboot the pi the screen comes back on again.

Looking at the octoprint.log I am seeing errors like these and I am at a loss to work out what is causing the issue:

2020-12-24 22:44:20,500 - octoprint.util.comm - INFO - Pausing/resuming job on behalf of plugin display_panel
2020-12-24 22:44:20,501 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-12-24 22:44:22,731 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: BSM/Pass_Through_Connector/VC_Body.gcode, owner: mini, user: None
2020-12-24 22:44:22,805 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-12-24 22:44:36,717 - octoprint.util.comm - INFO - Pausing/resuming job on behalf of user mini
2020-12-24 22:44:36,719 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-12-24 22:44:36,742 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: BSM/Pass_Through_Connector/VC_Body.gcode, owner: mini, user: mini
2020-12-24 22:44:36,806 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-12-24 22:45:02,791 - octoprint.plugins.display_panel - WARNING - An exception of type OSError occurred on log_error. Arguments: (121, 'Remote I/O error')
2020-12-24 22:47:58,131 - octoprint.util.comm - INFO - Pausing/resuming job on behalf of plugin display_panel
2020-12-24 22:47:58,131 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-12-24 22:48:00,879 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: BSM/Pass_Through_Connector/VC_Body.gcode, owner: mini, user: None
2020-12-24 22:48:00,946 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-12-24 22:48:00,972 - octoprint.plugins.display_panel - WARNING - An exception of type OSError occurred on log_error. Arguments: (121, 'Remote I/O error')
2020-12-24 22:48:01,004 - octoprint.plugins.display_panel - WARNING - An exception of type OSError occurred on log_error. Arguments: (121, 'Remote I/O error')
2020-12-24 22:48:06,028 - octoprint.util.comm - INFO - Pausing/resuming job on behalf of user mini
2020-12-24 22:48:06,029 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-12-24 22:48:06,051 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: BSM/Pass_Through_Connector/VC_Body.gcode, owner: mini, user: mini
2020-12-24 22:48:06,118 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2020-12-24 22:48:06,196 - octoprint.plugins.display_panel - WARNING - An exception of type OSError occurred on log_error. Arguments: (121, 'Remote I/O error')
2020-12-24 22:48:06,260 - octoprint.plugins.display_panel - WARNING - An exception of type OSError occurred on log_error. Arguments: (121, 'Remote I/O error')
2020-12-24 22:48:06,314 - octoprint.plugins.display_panel - WARNING - An exception of type OSError occurred on log_error. Arguments: (121, 'Remote I/O error')

sethvoltz, 2020-12-24:
Sounds like a wiring issue causing false detection of a button press. Check those wires for possible shorts or loose connection that could trigger on vibration.

bluespaniel, 2020-12-24:
Thanks for the quick response, I have already checked but will do again in the morning. I will try to change the routing of the wires in case any interference is a possible cause. I am not sure why the screen goes blank though.

sethvoltz, 2020-12-24:
That sounds like an i2c issue where the signal is coming in malformed which (IIRC) causes it to blank out. Others here have mentioned that some cheap wiring that came with their displays was flaky and caused all sorts of issues where the display would not be recognized, buttons would fail, etc.

Rewiring them with known good wires fixed those issues. Hopefully that works for you. Please let me know either way.

bluespaniel, 2020-12-24:
Thanks, I will do that. Awesome plugin by the way. 

I will try direct soldering onto the GPIO pins rather than using the jumper wire connectors to rule out vibration there.

bluespaniel, 2020-12-25:
Completely rewired the screen and switches soldering them onto the GPIO pins rather than using those cheap Chinese connectors and so far after two four hour prints no problem at all.

So it looks like it was a wiring issue as suggested so I will close this. Thanks for the help and thanks again for a great plugin.

closed 2020-12-25 by bluespaniel