sethvoltz/OctoPrint-DisplayPanel #10: Can not install plugin

Hobby4life, 2020-12-28:
After following the manual I still cant install the plugin.
It is also not visible in the plugin manager.

I did everything succesful following the installation guide.

I still get the following message in the plugin_pluginmanager_console log file:

2020-12-28 19:53:41,535 ! ERROR: Could not find a version that satisfies the requirement Adafruit-Blinka (from adafruit-circuitpython-ssd1306->OctoPrint-DisplayPanel==1.0.0) (from versions: none)
2020-12-28 19:53:41,536 ! ERROR: No matching distribution found for Adafruit-Blinka (from adafruit-circuitpython-ssd1306->OctoPrint-DisplayPanel==1.0.0)

sethvoltz, 2020-12-29:
A few others have opened this issue before. Please triple check that you have done the Python 3 upgrade according to the link in the README. This plugin _only_ works on Python 3 Octoprint on the Raspberry Pi. 

I'm not sure why Octoprint itself is not enforcing the Python version.

Hobby4life, 2020-12-29:
Finally got it to work :-)

In my case following the readme and istalling python3 on one of my pi's worked with some bypasses.

A real fresh install of octopi worked fine following the readme.

closed 2020-12-29 by Hobby4life