sethvoltz/OctoPrint-DisplayPanel #11: Display Z request

Hobby4life, 2020-12-29:
Hi. I would like to request a implementation of showing the Z position.

Like: https://plugins.octoprint.org/plugins/displayz/

sethvoltz, 2020-12-29:
If the data is available in the current data model then this is doable. The referenced plugin has a tooltip that suggests the value shown might be unreliable so I would like to find a better way.

unclej84, 2021-01-01:
Hi,

what about the [DisplayLayerProgress](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress) Plugin? It has an api which provides the height and layer numbers (current and total each). That might be a quite easy way to implement.

Regards Jörg

unclej84, 2021-01-16:
I had a look at this when working on better progress values and the DisplayLayerProgress plugin sends events on layer or height changes. So these could be used to display the Z height during prints. I am not sure where to put this information on the screen.

unclej84, 2021-01-24:
I think with the layout adaptions for dual color displays there should now be enough space for the additional line. So what shall be displayed? Position of toolhead/hotend or height of current layer in Z in mm (e.g. `Height:        10.4 /  54.6 mm`)? Or do you want the layer number (e.g. `Layer:     4 /  503`)?

Andy-ABTec, 2021-01-24:
Woh! - too many options,...

My money goes with Z in mm and layers selectable via settings - sorry!

unclej84, 2021-01-24:
So no height or just fixed mm?

bluespaniel, 2021-01-24:
Could you squeeze both in? Layer x of y and height in mm?

unclej84, 2021-01-24:
I prepared something but could not test. If you want to try [here](https://github.com/unclej84/OctoPrint-DisplayPanel/tree/feature-displaylayerprogress/octoprint_display_panel)?It requires DisplayLayerHeight plugin to work - displayZ plugin does not provide any information - just adds some javascript to the browser. Please not that you have to copy both `__init__.py` and `templates/display_panel_settings.jinja2` for testing. Maybe the texts must be adjusted in order to fit into one line.

unclej84, 2021-01-24:
After a simple test on console I had to set the content to `layer/totalLayer;height/totalHeight`. Unit for height or any letters did not fit when layer count is 4digits and height including decimal places 5.

Andy-ABTec, 2021-01-24:
If you're not happy with it go with  Z mm/Z-Max mm  OR Layer/Layer-max selectable in the settings menu, then the user has the choice - there's only so much you can squeeze onto that tiny display and keep it readable!

unclej84, 2021-01-24:
Had the chance to test it a bit - this is the current status:
[image]

bluespaniel, 2021-01-24:
Very nice

unclej84, 2021-01-24:
Thanks, would be interesting to know what the others think about it? Especially @Hobby4life as he opened the issue.

sethvoltz, 2021-01-25:
_THIS_ is why I love this community! You folks rock and I am loving these mods! I haven’t had time to review the latest PRs (sorry, home and work life are busy) but I’ll get to them, promise. 

I am digging the shorter progress bar and the extra line of text, especially as it gives a spot for the z-height, which I didn’t really want to add another screen mode for.

sethvoltz, 2021-01-30:
Merged, look for it in v1.4.0

closed 2021-01-30 by sethvoltz