sethvoltz/OctoPrint-DisplayPanel #13: Organisation of settings panel

unclej84, 2021-01-02:
Hello,

as the base for the settings panel is prepared in #4 and new setting shall be added according to #5 perhaps it would be nice to organize the panel a bit more. I am thinking of tabs "Buttons" and "Display" with following options:

Buttons: Pin GPIO setting, debounce time
Display: display address, rotation, date format, timeout time, timeout selection (timeout on printer disconnected, connected but idle, or always)

The bottom links could then be placed on the related tab.
What do you think about that?

Andy-ABTec, 2021-01-02:
I'm happy to go with your suggestion for now (and whatever others think), but does the plugin configuration popup actually **need** a title, people should know where they are from the plugin list on the left-hand side of the same popup, and should it be "**_Display Layer Plugin_**" or just "**_Display Plugin_**"? I'm happy with either, but it needs to be consistent - just a couple of thoughts...

sethvoltz, 2021-01-02:
I just merged the config panel - We can continue here or within a new PR

unclej84, 2021-01-03:
PR is in preparation. Changed the title to match the name in the plugin list.

unclej84, 2021-01-03:
This is how it looks with tabs:

[image]
[image]

Andy-ABTec, 2021-01-03:
Nice!

Andy-ABTec, 2021-01-03:
Sorry, this might be one for @sethvoltz following on from the initial registration, but I've just re-installed the plugin from scratch and noticed that when uninstalled the Plugin Manager calls it the "**OctoPrint Display Panel**", but once it is installed its called "**Display Panel Plugin**" - _confusing_ 🤔

Andy-ABTec, 2021-01-07:
Talking of confusing - any thoughts about this?
[image]

Andy-ABTec, 2021-01-07:
Looking around I note that some plugins use GPIO.BCM and some use GPIO.BOARD. Is there a preference? Should both be supported (or are they already?) Is it just a matter of personal taste?

Sorry for all the questions, I'm still finding my way around the Pi architecture...

sethvoltz, 2021-01-08:
Oh no, I hadn’t realized someone had the name first. This started as a hack so I didn’t look too far. I’m open to renaming as I came later. Ideas for a different name?

Andy-ABTec, 2021-01-08:
Mini Panel, Micro Panel - I don't know, something that infers a small display...

sethvoltz, 2021-01-14:
Hoping to have some time this weekend to get into this. Mini or Micro Panel has a nice ring to it.

unclej84, 2021-01-14:
If you could check the tab PR I would directly add the timeout PR (based on the tabs in settings page) - have been testing for around a week now.

sethvoltz, 2021-01-14:
Sure thing! Laptop is away for the evening but I’ll take a look soon. Thanks for the reminder!

sethvoltz, 2021-01-16:
@unclej84 or @Andy-ABTec do either of you know if this is sufficient for renaming a plugin? I left the identifier the same so folks don't have to uninstall and reinstall the new version

unclej84, 2021-01-16:
Sorry, I have no idea how this actually works but I think as you changed the name the overlays feature needs to be used. Best would be to get in contact with Gina or jneilliii

Andy-ABTec, 2021-01-16:
Likewise, I have no idea...

sethvoltz, 2021-01-16:
Opened this thread if you want to follow along https://community.octoprint.org/t/how-to-cleanly-rename-an-octoprint-plugin/29155

Andy-ABTec, 2021-01-17:
@sethvoltz, not sure if you're aware but OctoPrint isn't offering the 1.2.0 upgrade as available (for me at least, I might be missing something) Plugin Manager says I'm running 1.1.0 - related to the name change maybe?

unclej84, 2021-01-17:
I got it offered and installed this morning. Have you tried to force an update? Maybe you hit the ratelimit on github?

Andy-ABTec, 2021-01-17:
I "piped" it over the top of the existing installation and all seems ok, now saying 1.2.0, just wait and see on the next release...

sethvoltz, 2021-01-17:
I’ve been trying to figure that out myself, I’ve only gotten 1.1.0. I followed all the points from the bootstrap app that were mentioned in the docs so I’m thinking the plugin site periodically polls to see if there’s a newer version and then advertises it?

Andy-ABTec, 2021-01-17:
Confusing or what... as previously mentioned the only way I got the later version was to force it over the top with pip, might be worth leaving it for a while and then fake a new release just to see what happens.

Other than that I guess it back to Gina or jneilliii...

sethvoltz, 2021-01-28:
Thanks for the feedback and PRs! I got offered the latest version a few days ago. I'll be cutting a release with the latest changes in a bit, should be notifying folks in a day or two.

closed 2021-01-28 by sethvoltz