sethvoltz/OctoPrint-DisplayPanel #14: Case design

Andy-ABTec, 2021-01-02:
Just in case (no pun intended) any Fusion 360 users want to take a poke at my case design before I upload it to Thingiverse here's a Fusion archive for you to play with... 😎

[oled case.zip](https://github.com/sethvoltz/OctoPrint-DisplayPanel/files/5760427/oled.case.zip)

[image]

sethvoltz, 2021-01-02:
Have you tried printing it yet? There doesn't appear to be a tolerance gap between the two parts in that screenshot

Andy-ABTec, 2021-01-03:
You mean like this? 😎

[image]

First prototype, printed at 0.16mm, it was an experiment with "Z Hop" in Cura to see how well I could "insert" the text in a different colour by changing the filament and overprinting (white first, then black). I was pretty pleased when I took it off the bed.

The tolerance thing was deliberate on my part, I wanted a friction fit that didn't keep falling apart. The detent in the base lets you apply an initial tweak with a knife to open it again.

The width and depth were specific to allow mounting across 40mm  V rail or along 20mm V rail using with the printed "T nut" add-ons. If it's mounted from the back the cabling can still exit the back and run up or down in one of the rails - reminds me I need to find some snap-in covers for the slots...

bluespaniel, 2021-01-03:
Nice case, I incorporated the screen and buttons into the skirt of my Voron 0.
[image]

Andy-ABTec, 2021-01-03:
That's cool...

bluespaniel, 2021-01-03:
Thanks, I had just enough room to fit the screen in. I did it before it could be rotated 180 degrees so I had to bend the pins to keep them away from the PSU. With the new version of the plugin, the installation would be much easier.

Andy-ABTec, 2021-01-03:
@sethvoltz it might be an idea to leave this thread open as somewhere for folks to share their designs with others and generate new ideas...

sethvoltz, 2021-01-14:
I like the thread idea. These case prints are great, I am so happy folks are modding and extending the design and plugin!

sethvoltz, 2021-01-28:
I just enavbled Github Discussions for this repo. I wanted to convert this issue over to a discussion as the documentation suggests I can, however there were apparently bugs discovered over the last couple months and that feature was disabled. I'd love to seed these builds in a new discussion thread, though, so I created a show and tell thread and will be closing this one out.

https://github.com/sethvoltz/OctoPrint-DisplayPanel/discussions/32

@Andy-ABTec and @bluespaniel would you mind posting your builds on there to show off your cool mods?

Thanks!

closed 2021-01-28 by sethvoltz