sethvoltz/OctoPrint-DisplayPanel #15: GPIO Pins...

Andy-ABTec, 2021-01-02:
Carrying on the "conversation" from https://github.com/sethvoltz/OctoPrint-DisplayPanel/issues/4 (and bearing in mind I'm not particularly familiar with the Pi...)

It seems that when the Pi first boots, all GPIO pins are set to input. GPIO's 0 to 8 have pull-ups (to 3.3V) enabled - the rest have pull-downs (to ground) enabled. How this is changed I have no idea, but I would imagine that there is a Python library that handles it.

The reading I have done implies that these pull-ups and pull-downs are generally sufficient if the pin in question isn't being used, however connecting them to switches, buttons and the like can lead to problems in electrically noisy environments (like a 3D Printer?) even with sensible wiring routing and screening.

The following circuit is commonly used to suppress spurious triggering: -

[image]

The 1K resistor "nails" the GPIO pin high and the 100R resistor and 100nF capacitor form a low pass filter with a cut-off frequency of just under 16KHz (1/2πRC) to help deal with noise pickup.

The circuit should be placed as close to the Pi as possible.

sethvoltz, 2021-01-02:
Thanks for this. This seems like it would be a great "Printer PHat" idea, where a bunch of the GPIO can have this baked in with pins and/or screw terminals to connect out to peripherals like this display.

Andy-ABTec, 2021-01-03:
That's what I was thinking, it would have to be a little bit "DIY" because not everyone's use would be the same, so I guess a general style breakout board that can be hard-wired for output or input with the filter would be the way to go - should be doable, the hardest part is figuring out if/how I can squeeze it between a Pi and another hat. (The Pi on my printer actually runs from a PoE hat.)

Andy-ABTec, 2021-01-07:
Just found a low profile 40 pin pass through connector as used in pi hats, so it looks like a board may well be on the cards

Andy-ABTec, 2021-01-10:
This sort of thing?  Wonder If I can fit it all on a Pi sized board? Not used to this new fangled SMD stuff (yet!) 🤔

[Raspberry Pi GPIO.pdf](https://github.com/sethvoltz/OctoPrint-DisplayPanel/files/5792859/Raspberry.Pi.GPIO.pdf)

Good to do some design again, it's been a few years - like 20! My first schematic in Fusion 360 as well...

sethvoltz, 2021-01-14:
@Andy-ABTec I like where you’re going with this!

How was the Fusion360 experience? I tried a few times in the early beta days and it is all Greek to me. The last times I designed PCBs I hand drew them for a school project. As for SMD, with good solder mask and flux, I find them easy enough to solder. Just double check the pad footprint in CAD is correct before you send it to the board house, haha.

Andy-ABTec, 2021-01-14:
@sethvoltz, It's not going any further just now...

I was expecting Fusion Electronics to be much more polished than it is, and the Schematic Editor lulled me into a false sense of security. I was doing reasonably well allowing for the learning curve until I switched to the board editor - then things started to fall apart. The board editor is horrible and confusing, I'm sure it will get better, but at the moment it stinks.

As for creating your own libraries, I've spent 3 days trying to create a custom library containing the 40 pin SMD socket so I can use it for the board - still no joy. The documentation seems to be nonexistent, wildly inaccurate or written for Eagle before Autodesk started fiddling with it...

As for the actual board, I'm hoping not to need a production house, I've nearly finished building one of these:

[image]

[The Ant Desktop PCB Mill](https://www.youtube.com/watch?v=O7gI6XYmP2o)

When I was an apprentice we used to make our own boards the old fashioned way, tape and film, photographic reduction to size, silkscreen, etch resist and on to the ferric chloride bath...

Building "The Ant" this winter is one of my distractions from losing my wife to heart failure around 18 months ago...

Andy-ABTec, 2021-01-29:
@sethvoltz you want to move this thread to discussions?

sethvoltz, 2021-01-29:
I was hoping to wait for GH to fix the bugs and let me move it automatically but, yeah, if you wanna kick it off over in a new discussion I can close this one out.

Andy-ABTec, 2021-01-29:
Fine, I've got a local copy of this thread... I'll probably split it into a debounce discussion and a hat discussion, but I have everything I need from here

sethvoltz, 2021-01-29:
Thanks Andy. I like the idea of splitting it into two conversations as well, especially if the pHat idea takes off, you could be forking that off to your own repo 😄

Andy-ABTec, 2021-01-29:
That was my thought, I just need to get over the Fusion Electronics Library mess - working on it but really painful...

sethvoltz, 2021-01-30:
Since this is going over on Discussions, closing this out.

closed 2021-01-30 by sethvoltz