sethvoltz/OctoPrint-DisplayPanel #17: Plugin installation problems

hallskelet, 2021-01-03:
I connect display to Orange Pi board. And I checked **sudo i2cdetect -y 0** 
` orangepi@orangepipc:~$ sudo i2cdetect -y 0
[sudo] password for orangepi:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
`
the display is defined. I install a plugin and it doesn't appear in the plugins list.
How do I fix this problem?

hallskelet, 2021-01-03:
[image]
after reboot:
[image]

unclej84, 2021-01-03:
The current version in the plugin repo does not have any settings item. So you won't see it in the list on the left side but it should be listed under Plugin Manager. When everything is connected properly the display should be active and at least the mode button working.

hallskelet, 2021-01-03:
I have record in log file: "octoprint.plugin.core - ERROR - Error loading plugin display_panel ImportError"

I think this problem is related to incorrect work of adafruit libgpiod.
The problem is that I am not a programmer, and I do not understand where the error should be corrected in the library.
I am now trying to find an instruction to fix the fact that this library is called only through SUDO. But I haven't found a solution yet

hallskelet, 2021-01-03:
I don’t know how, but I got the code
```
    import board
    import digitalio
    import busio
     
    print("Hello blinka!")
     
    # Try to great a Digital input
    pin = digitalio.DigitalInOut(board.PA6)
    print("Digital IO ok!")
     
    # Try to create an I2C device
    i2c = busio.I2C(board.SCL, board.SDA)
    print("I2C ok!")
     
    print("done!")

```
gave this result without "sudo"
```
 python3 blinkatest.py                                    
Hello blinka!
Digital IO ok!
I2C ok!
done!

```

hallskelet, 2021-01-03:
But I still have an error:

```
2021-01-03 22:12:25,883 - octoprint.plugin.core - ERROR - Error loading plugin display_panel
ImportError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint_display_panel/__init__.py", line 16, in <module>
    from board import SCL, SDA
  File "/home/orangepi/oprint/lib/python3.7/site-packages/board.py", line 76, in <module>
    from adafruit_blinka.board.orangepi.orangepipc import *
  File "/home/orangepi/oprint/lib/python3.7/site-packages/adafruit_blinka/board/orangepi/orangepipc.py", line 3, in <module>
    from adafruit_blinka.microcontroller.allwinner.h3 import pin
  File "/home/orangepi/oprint/lib/python3.7/site-packages/adafruit_blinka/microcontroller/allwinner/h3/pin.py", line 2, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/home/orangepi/oprint/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 8, in <module>
    ) from ImportError
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh
```

unclej84, 2021-01-03:
Octoprint uses a virtual env which might contain different packages than your user account. Do you have a venv directory somewhere in your home directory for octoprint? e.g. oprint/venv?

unclej84, 2021-01-03:
I just had a look at the setup script in your link under "[Install libgpiod](https://learn.adafruit.com/circuitpython-on-orangepi-linux/orange-pi-r1#install-libgpiod-3029538-23)". The last 3 lines in the script copy the files to your system python directory. You could try to change `/usr/local/lib/python3.?/dist-packages/` by `/home/orangepi/oprint/lib/python3.7/site-packages/` and then rerun the script. This should copy the required library to octoprints python directory. Hope this works.

hallskelet, 2021-01-03:
yes. i have "oprint" dir.
before configuring python i used the command
"source oprint/bin/activate".

as a result, I have:

```
2021-01-03 23:56:07,526 - octoprint.plugin.core - ERROR - Error loading plugin display_panel
Traceback (most recent call last):
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint_display_panel/__init__.py", line 16, in <module>
    from board import SCL, SDA
  File "/home/orangepi/oprint/lib/python3.7/board.py", line 76, in <module>
    from adafruit_blinka.board.orangepi.orangepipc import *
  File "/home/orangepi/oprint/lib/python3.7/adafruit_blinka/board/orangepi/orangepipc.py", line 3, in <module>
    from adafruit_blinka.microcontroller.allwinner.h3 import pin
  File "/home/orangepi/oprint/lib/python3.7/adafruit_blinka/microcontroller/allwinner/h3/pin.py", line 4, in <module>
    PA0 = Pin(0)
  File "/home/orangepi/oprint/lib/python3.7/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 34, in __init__
    self._chip = gpiod.Chip("gpiochip0", gpiod.Chip.OPEN_BY_NAME)
AttributeError: module 'gpiod' has no attribute 'Chip'
```

unclej84, 2021-01-03:
according to [this](https://github.com/adafruit/Adafruit_Blinka/issues/328) post you might be running the script using legacy option usibg `-l` or `--legacy`

hallskelet, 2021-01-05:
Thanks for the help. I wrote about the problem on that thread. maybe they will advise me something. Because I don't even understand where these scripts are incorrectly installed.

sethvoltz, 2021-01-14:
@hallskelet any update here? This sounds like issues others have had where Octoprint was still running on Python 2 and not upgraded to Python 3 fully. The GPIO libraries failing is the first sign as they do not work at all on Python 2.

sethvoltz, 2021-01-16:
Closing this issue for now. If you have installation issues that happen after you confirmed Python 3 upgrade, please re-open.

closed 2021-01-16 by sethvoltz

hallskelet, 2021-01-17:
I have a python 3 installation and the "WiringOP" libraries are working fine. The problem is precisely in the incorrectly working "adafruit_blinka" library, perhaps this library somehow does not work correctly with Armbian. But my knowledge is not enough to understand why the error appears. Is it possible to replace the call of this function with a call from "WiringOP" or just specify the pin numbers for SPI0 with constants?

Pele79, 2021-02-09:
I am having the same issue on an RPi4 4gb version on a fresh install of the latest Octopi release, I did have is on the previous version which I had updated to Python 3 manually, but when I couldn't get it to work I decided to try a fresh install to no avail.  If anyone has a fix that would be greatly appreciated.