sethvoltz/OctoPrint-DisplayPanel #19: GUI on dual color screen

jghake, 2021-01-13:
The display on screens with dual color is cut in half.
[image]

sethvoltz, 2021-01-14:
First off, that green filament is bomb 💣 , where did you get it?

As for the split color, I definitely didn’t design it for use with those. The layout could be modified but you’d likely lose a line of text. Just spitballing, maybe the progress bar would move to the top and three lines could be shown below? I don’t have any plans to do this but if you submit a PR with the layout and a config checkbox, I’ll review it for addition.

jghake, 2021-01-14:
I don't know much about these but I can/will look in to it eventually. Right now the buttons don't work and I suspect a wiring problem.

This filament is my favorite color. 3DSolutech Camo Green.
https://amazon.com/3D-Solutech-Filament-Dimensional-Accuracy/dp/B07BN8H2LQ/

unclej84, 2021-01-17:
Maybe this can be realized with an offset to the blocks for top and bottom (e.g. int(bool Switch)*(+/-)offset). That would be easy to handle. Could you provide the information how many pixels the top area has in height (16?)?
Then I will have a look at it and prepare a PR.

jghake, 2021-01-18:
I tried writing the code myself and all I managed to do was break my code... I'm definitely not a programmer. 

As far as pixel count I guess it's probably 16. My eyes can't see it, and Google has failed me.

rbolinger, 2021-01-22:
I would love to see a configuration option for progress bar at top or bottom, I have a similar OLED display and would opt to put the progress bar at the top given the choice. And great plugin, amazing job!

Don't know how many pixels the yellow portion is but here is the OLED module on Amazon https://www.amazon.com/gp/product/B072Q2X2LL

unclej84, 2021-01-22:
The description says 2 rows yellow and 6 rows normale - so 8 Pixels per row makes 16. I can try to make a test version in my repo but cannot test right now - if you could change the files manually for testing afterwards I will do this quickly

unclej84, 2021-01-22:
is this only relevant for the printing screen or do the other two also need to be altered?

jghake, 2021-01-23:
I uploaded the files, set the option in the plug-in preferences and rebooted. This is what it looks like now. When I press m the screen changes to the blank bottom, press a few more times and you get the one line chopped off. 

No job
[image]
[image]

Printing
[image]
[image]

unclej84, 2021-01-23:
I updated the `__init__.py` - the progress bar area cleared too much of the screen. I also made some minor positioning changes to fit the colored area. Could you try again please?

unclej84, 2021-01-23:
Could you please also check the normal layout with progress bar on bottom (photos would be really helpful) to see if the positioning needs to be different or if it is okay for this too?

unclej84, 2021-01-23:
Could you please also try `rotation` set and normal layout (if it is okay now)? Would be interesting to see how this would be when having the colored display rotated. A photo would be appreciated.

jghake, 2021-01-23:
Looks a lot better. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

unclej84, 2021-01-23:
So the progress bar should be 2 pixels higher I guess. Let me check this quickly. Sorry for the confusion regarding rotation - i meant rotation without progress on top - wanted to see if this messes the "normal" layout.

jghake, 2021-01-23:
Better!

[image]
[image]
[image]
[image]

unclej84, 2021-01-23:
okay, I made a final tweek because the last line was printed on top - now it should fit. will prepare the PR. If you want you can check the final file before @sethvoltz reviews the PR

jghake, 2021-01-24:
[perfect](https://media.tenor.com/images/47f86b14d1742ff3531cf0db3b3f9f70/tenor.gif)
[image]

unclej84, 2021-01-24:
Yeah, looks nice. I had another look at your photos with `progress on top` and `rotation` where the last line is cut a bit so I added an additional pixel as offset to move this line completely to the colored area without cutting the top (main branch in my repo + PR here). Think we got it now.

sethvoltz, 2021-01-28:
Merged in fix, will be cutting a release soon

closed 2021-01-28 by sethvoltz