sethvoltz/OctoPrint-DisplayPanel #20: Progress bar / percentage

unclej84, 2021-01-14:
Hello,

I noticed that the progress used is not very helpful as it uses octoprints internal progress which is based on transferred file size / bytes. In my case it is before even starting to home (during heat up) already jumped to 10%. For testing proposes I changed it from provided progress by octoprint with `printTimeLeft / (printTimeLeft + printTime) * 100`. This gives much better results and seems to be the method to be used also by other plugins like Dashboard or DisplayLayerProgress.
What do others think?

Regards Jörg

sethvoltz, 2021-01-14:
I use PrintTimeGenius which overrides the estimate but iirc the estimate source is labeled so there could for sure be some better logic to switch calculation method based on presence or absence of those extensions. 

I like the idea of calculating off end time for the inaccurate estimate sources. It’d probably time I add back in some of the debug code I started with and properly tag it as Debug level for gathering this data later on.

unclej84, 2021-01-14:
I use PrintTimeGenius too and as far as I understood its "total time" is played back to octoprint internal `printTime`. So either the internal total time is provided or if the plugin is installed you directly get the "corrected" total time and this is used.

closed 2021-01-22 by unclej84