sethvoltz/OctoPrint-DisplayPanel #24: Jobs cancelling

jghake, 2021-01-19:
I've been struggling to print lately as my jobs will randomly stop in the middle of a print. I've been chasing around camera issues thinking that was the issue but I was just trying to get a linear advance test printed and the job stopped halfway through. Looking through the logs I found this.

2021-01-19 09:52:18,093 - octoprint.util.comm - INFO - Starting job on behalf of user jghake
2021-01-19 09:52:18,101 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2021-01-19 09:52:18,109 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: TTYT3dkfactor.gcode, owner: jghake, user: jghake
2021-01-19 09:52:18,173 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-01-19 09:52:18,183 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2021-01-19 09:52:18,354 - octoprint.plugins.DisplayLayerProgress - INFO - Printing started. Detailed progress started.{'name': 'TTYT3dkfactor.gcode', 'path': 'TTYT3dkfactor.gcode', 'origin': 'local', 'size': 20983, 'owner': 'jghake', 'user': 'jghake'}
2021-01-19 09:52:18,358 - octoprint.plugins.PrintJobHistory - INFO - Try reading Temperature from PreHeat-Plugin...
2021-01-19 09:52:18,395 - octoprint.plugins.PrintJobHistory - INFO - ... Temperature found '60.0' ' Tool 'tool0' '220.0'
2021-01-19 09:52:19,001 - root - ERROR - {"remote_status": {"should_watch": true}, "type": "printer.message"}
2021-01-19 09:52:19,054 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': 'local', 'file': 'f1d8fc619f172a83a0bbfc7f48964f4f8f55a4d9'}
2021-01-19 09:56:10,590 - octoprint.util.comm - INFO - Cancelling job on behalf of plugin display_panel
2021-01-19 09:56:10,593 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2021-01-19 09:56:10,618 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: TTYT3dkfactor.gcode, owner: jghake, user: None
2021-01-19 09:56:11,008 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2021-01-19 09:56:11,019 - octoprint.plugins.PrintJobHistory - INFO - Print result:canceled, CaptureMode:always
2021-01-19 09:56:11,019 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2021-01-19 09:56:11,030 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-01-19 09:56:11,822 - octoprint.plugins.tracking - INFO - Sent tracking event print_cancelled, payload: {'origin': 'local', 'file': 'f1d8fc619f172a83a0bbfc7f48964f4f8f55a4d9', 'elapsed': 232}
2021-01-19 09:56:12,085 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-01-19 09:56:12,090 - octoprint.plugins.PrintJobHistory - WARNING - Thumbnail not found in print metadata
2021-01-19 09:56:12,091 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try taking snapshot '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210119-095218.jpg' from 'http://192.168.45.88:8080/?action=snapshot'
2021-01-19 09:56:12,097 - octoprint.plugins.PrintJobHistory - INFO - Empty filamentModel, because Filamentmanager not installed!
2021-01-19 09:56:12,171 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Process snapshot image
2021-01-19 09:56:12,495 - octoprint.plugins.DisplayLayerProgress - INFO - Printing stopped. Detailed progress stopped.
2021-01-19 09:56:12,577 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Image stored to '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210119-095218.jpg'
2021-01-19 09:56:12,761 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"

I was nowhere near the printer when this happened so I'm not sure why the job was stopped on behalf of display panel.

sethvoltz, 2021-01-20:
Other users have seen this if they use really long cables between the buttons and the Pi or if you have loose wires, usually cheap ones that weren’t crimped well or don’t fit the Pi headers well. This can cause vibration on the printer to wiggle the cables or electrical interference causing false button presses. 

Please check your wiring to make sure the connections are tight. For electrical interference due to long cables, you might be able to debounce with a capacitor across the connections close to the Pi, but I can’t offer advice there as I haven’t had to do it.

jghake, 2021-01-20:
I can check all the connections again. The wire isn't long, maybe 12". I used cat5e cable with Dupont connectors crimped on the end.

Andy-ABTec, 2021-01-20:
Out of interest did you use Cat5e patch or installation cable? individual wires, or twisted pairs for each switch?

Might be worth taking a look at [[Discussion] Hardware debounce GPIO Pins #15](https://github.com/sethvoltz/OctoPrint-DisplayPanel/issues/15), you'll find a debounce circuit there...

I'm currently working on a low profile "Breakout Hat" providing proper header pins for to the I²C bus and  debounced GPIO but at the moment the Fusion Electronics Library is not best my friend, I've been playing for ages trying to create a 40 pin SMD socket to allow the board to stack - but no joy yet...

jghake, 2021-01-20:
It's solid core riser cat5e. I haven't tried disconnecting anything yet. 

That's cool you're making a hat! I would definitely interested in one.

jghake, 2021-01-21:
I've moved the pi off the buck converter connected to the Ender power supply and to a USB charger. I haven't seen a job cancel yet, hopefully it stays that way. 

Would a press and hold for 3 seconds confirm instead of a second press eliminate issues?

sethvoltz, 2021-01-28:
@jghake glad to see the issue cleared up. Maybe there was some power ripple from the buck converter bleeding into the switch signal lines? 

I personally prefer the press-twice but PRs are welcome to add the feature. All I ask is it is a configuration option and I’ll work with you to get it mergable.

closed 2021-01-28 by sethvoltz