sethvoltz/OctoPrint-DisplayPanel #25: Add ability to change temperature when not printing anything.

rbolinger, 2021-01-22:
I love that this plugin and associated hardware allows for controlling OctoPrint from the printer, better form factor than putting a full touch screen display by the printer. Would love to be able to control the temperature from it too, something like a mode selection for temperature control. Then press play to select that and mode through the presets and pressing play again would select it.

Not having to go to my laptop and set the hotend temperature for a filament change would be nice. Would be happy to donate the proverbial coffee.

unclej84, 2021-01-22:
Could you describe a bit more how the workflow / configuration should look like? Switching mode to temperature screen ("M"), toggle bed / hotend using ? (which key), setting temperature + / - or presets? How to confirm or abort/cancel?

rbolinger, 2021-01-25:
Press Mode to flip through the modes landing on the new change temperature.
-Press Play to select into that mode.
--Optionally have another mode selection to select setting tool or bed temperature and potentially multiple tools.
---Show list of OctoPrint temperature presets for selected Too or Bed with an arrow or something to show what is selected.
---Pressing Play would select that temperature and set it accordingly, also return to primary primary mode.
-Pressing cancel at any time would exit out of the temperature set mode and back to primary display mode.

And maybe could replace the status bar/waiting for job section with hints for M=Change, Play=Select, Cancel

sethvoltz, 2021-02-28:
Thanks for this description. I don't see adding this myself as I prefer the additional control of OctoPrint's UI for complex changes or the tuning menu on my printer's display for fine tuning during a print. If you would like to submit a PR with the functionality, I think your use of the buttons fits well enough and we can discuss the specifics over the code itself. Thanks!

closed 2021-02-28 by sethvoltz