sethvoltz/OctoPrint-DisplayPanel #26: Using I2C0 instead of I2C1

rbolinger, 2021-01-22:
Anyone know if this will work on the I2C0, pins 27 and 28, rather than I2C1, pins 3 and 5? I am trying to add a power button too and while I can splice the SCL from the OLED display into a momentary switch running a script to watch for pressing a button on GPIO3 (pin 5) kills the communication to the display. And yeah I could move the momentary switch to another GPIO but as far as I know the wake up pi only works when you short GPIO3 to a ground.

Right now I have it still connected to I2C1 and the momentary switch spliced, it allows me to wake the pi from a shutdown state but not shut it down from the button.

unclej84, 2021-01-22:
I am also using a button between pin 5 and ground - just made a small adapter splitting the pin 5 socket to 3 pins.
You have to try - change the `__init__.py` file [here](https://github.com/sethvoltz/OctoPrint-DisplayPanel/blob/4adc500f4d887924753c4caefbf4e8c4539f9365/octoprint_display_panel/__init__.py#L376)  and replace `SCL` by `D1` and `SDA` by `D0`. Afterwards restart octoprint and see if it works.

unclej84, 2021-01-22:
Please check [this](https://www.engineersgarage.com/raspberrypi/articles-raspberry-pi-i2c-bus-pins-smbus-smbus2-python/) before trying - seems other things have to be considered too. So on your own risk

rbolinger, 2021-01-25:
Thanks, I'll give that a try.

closed 2021-03-04 by sethvoltz