sethvoltz/OctoPrint-DisplayPanel #27: Updates to the plugin template

Andy-ABTec, 2021-01-23:
@sethvoltz I've been "fiddling" with the plugin template... Nothing major just fixed some grammar and tweaked some text to help with the readability (hopefully).

You can find it here: [https://github.com/Andy-ABTec/OctoPrint-DisplayPanel](https://github.com/Andy-ABTec/OctoPrint-DisplayPanel)

Please help yourself if you think it of use...

unclej84, 2021-01-23:
Have you checked the output of the quotes in line 149? They are showing fine with just `"`. Is there any reason escaping them?
Could you perhaps check my feature branch for progressOnTop? There is a new option for having the progress bar on top and it would be great to directly have "good" wording when creating the PR.

sethvoltz, 2021-01-23:
@Andy-ABTec you can also open a PR against this repo from yours instead of opening an issue and we can have comments and reviews directly on your code

Andy-ABTec, 2021-01-23:
OK, I'll take a look - didn't open a pr against this repo 'cos I'm still fairly new to Git/GitHub/GitHub Desktop and didn't want to screw any live stuff up!

Andy-ABTec, 2021-01-23:
There was no reason for escaping stuff other than I'm making it up as I go along.
I know my way around bourn/bash/HTML/CSS and to a lesser extent javascript (I was a Unix/Linux admin for 20 years and worked in electronics for 20 years before that!) I know my way around hardware, I designed enough of it in the past, but when it comes to code I'm a straight up/down Fortran guy, this OOP stuff leaves me cold!

Andy-ABTec, 2021-01-23:
@unclej84 I think if it was me I'd say something like "Show the Progress Bar at the top of the display" rather than "Display progress on top of printing screen".
GitHub doesn't seem to want me to fork your repo, I think because I've already forked Seths, but as I said I'm new to this and get lost in the jargon easily, so it may be something I'm doing or not doing...

unclej84, 2021-01-23:
Thanks for the proposal - I have incorporated it directly. PR is placed - now @sethvoltz has the fun to fit everything together

sethvoltz, 2021-01-28:
Merged in language changes and progress on top PRs.

closed 2021-01-28 by sethvoltz