sethvoltz/OctoPrint-DisplayPanel #38: Texts will be cut

NilsRo, 2021-01-29:
Hi,

at first I like to thank you for the cool Plugin. Put it on my Mega S to get additional information and control as the internal display is nit usable with Octoprint. Nice work. As the most important is said :-) now the small issue. The text in cut in my display on the first page on the right.  Not a real problem but I like you to know.

[image]

Cheers,
Nils

unclej84, 2021-01-29:
Hi,

looks like an issue with your display. Could you try to add `				self.draw.rectangle((0, 0, self.width, self.height), fill=255, outline=255, width=1)` (for tabs at the beginning) after [this](https://github.com/sethvoltz/OctoPrint-DisplayPanel/blob/ab934cf1cdcb7f8746634d60350cfe603b2e858e/octoprint_display_panel/__init__.py#L850) line (should be one tab to the left)? This should print a big white rectangle on the whole screen - if the pixels are still missing, then your display has a hardware defect.

NilsRo, 2021-01-29:
Hi.

By my mistake it is not the line across the display. This must be created by the refresh rate of the display. The ‚problem - if it could be named like this’ :-) is visible on the right. The row displays memory used does print the half of the percentage sign only. That’s similar to a second display I tried.

unclej84, 2021-01-29:
This issue is based on the length of the line - as you see the line with Load is also cut. A possible solution would be to shorten the static texts to just one letter (Load -> L, Mem -> M, Disk -> D, GB -> G and MB -> M). For testing you could update these in lines 730 to 733.

sethvoltz, 2021-01-30:
Yeah, I'm fine trimming those down. I'll add it to the next version

sethvoltz, 2021-01-30:
Look for a fix in v1.4.0

closed 2021-01-30 by sethvoltz

NilsRo, 2021-01-30:
That's better. Thanks a lot.