sethvoltz/OctoPrint-DisplayPanel #39: Add rotating information in large letters

NilsRo, 2021-01-29:
Hi.

As an enhancement it would be nice to add the possibility to rotate important information during print in large letters. The display used if fine but a little small to read the information shown if you do not stand directly in front of the printer.

Cheers,
Nils

sethvoltz, 2021-01-30:
Can you elaborate more what you're looking for?

NilsRo, 2021-01-30:
Hi. Something better readable (larger letters) with less information like with the simple LCD here: https://github.com/arttupii/Octoprint-Adafruit-LCD

Large letters with rotating information every 10seconds with time remaining, ETA,...

For me that is the information I am interesting in as a hobby user during print. I look to my printer and to know when the print is finished.

sethvoltz, 2021-02-28:
This plugin was designed to cram as much as possible into the screen as possible, but I'd be open to a mode that has larger text and fewer things, if you want to submit a PR.

I would also suggest that if you have a display on your printer already to check out the [Detailed Progress](https://plugins.octoprint.org/plugins/detailedprogress/) plugin. I use that as well to keep my printer's LCD up-to-date with print percentage which I can read across the room and then use my own monitor for more detailed stats close up.

sethvoltz, 2021-02-28:
I'm closing this for now, if you have a PR, please feel free to open it and we can continue the discussion over the code.

closed 2021-02-28 by sethvoltz