sethvoltz/OctoPrint-DisplayPanel #46: Can´t instal OctoPrint-Micro-Panel

dantosar, 2021-01-31:
Hi I have this issue:

Installiere Plugin...

Installiere Plugin von hochgeladenem Archiv.../home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpEXpK2c.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleProcessing /tmp/tmpEXpK2c.zipRequirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Micro-Panel==2.0.0) (1.5.3)Collecting adafruit-circuitpython-ssd1306Downloading https://files.pythonhosted.org/packages/3c/8e/f072f2f496ff1debe2f03158d92113aaec54ce7a701bf0696e96182bcc52/adafruit-circuitpython-ssd1306-2.10.1.tar.gz (40kB)Requirement already satisfied: pillow in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Micro-Panel==2.0.0) (6.2.2)Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.0.0)Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.10.4)Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2020.8.7)Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (5.8.0)Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.5.4)Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.5)Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.0.7)Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (5.4.1)Requirement already satisfied: feedparser<6,>=5.2.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (5.2.1)Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.10.9)Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.18.2)Requirement already satisfied: enum34<1.2,>=1.1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.1.10)Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.7.19)Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2.25.1)Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (7.1.2)Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.1.0)Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.4)Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2.11.2)Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.2)Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.5.0)Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.1.2)Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (4.0)Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2.8.5)Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.4.21)Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.5.0.1)Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (3.3.0)Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.57.0)Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.10.0)Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (5.1.1)Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (3.4)Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.1.1)Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (3.1.1)Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2019.8.19)Requirement already satisfied: chainmap<2,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.0.3)Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (3.7.4.1)Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.1.5.post0)Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2020.9.23)Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.19.5)Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.2.0)

Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (1.12.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (2.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Micro-Panel==2.0.0) (0.1)
ERROR: Could not find a version that satisfies the requirement Adafruit-Blinka (from adafruit-circuitpython-ssd1306->OctoPrint-Micro-Panel==2.0.0) (from versions: none)
ERROR: No matching distribution found for Adafruit-Blinka (from adafruit-circuitpython-ssd1306->OctoPrint-Micro-Panel==2.0.0)
Fehler!Could not parse output from pip, see plugin_pluginmanager_console.log for generated outputFehler!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

I´ve install manual over putty Adafruit-Blinka 

please help!

thanks

dantosar, 2021-01-31:
also check that i have i2c-tolls installed
[image]

Andy-ABTec, 2021-01-31:
Hello @dantosar, the plugin actually requires Python3 ( see [**Setup in the Readme**](https://github.com/sethvoltz/OctoPrint-DisplayPanel#setup)). By the look of the log you kindly supplied you still appear to be running Python2...

dantosar, 2021-01-31:
Thank you ! It was this! Thanks a lot

closed 2021-01-31 by dantosar