sethvoltz/OctoPrint-DisplayPanel #50: ENHANCEMENT? - Testing GPIUO pin assignment...

Andy-ABTec, 2021-02-04:
I've finally got the buttons for my tiny case yesterday (Damm Covid); and I was in the process of wiring everything, running wire in v-slot grooves with snap-in covers etc. when I had an idea...

How about adding a third tab called "Test" to the plugin configuration page, with nothing on the page but four boxes labelled "Mode", "Cancel", "Play" and "Pause". While the relevant button is pressed the associated box "lights up" (changes colour) and maybe we get beeps identifying which box was activated (1 beep, 2 beeps etc.) for the odd situation when you can't see the screen.

Good as a quick test of pin assignment and continuity?

Andy-ABTec, 2021-02-05:
Or you could maybe (probably better) add some form of indicator beside the button assignment drop downs on the first tab, so the assignment could be checked at the same time as testing without having to switch tabs...

sethvoltz, 2021-02-07:
I dig the idea. I'll leave it open here to see if anyone wants to take it up as a PR.

unclej84, 2021-02-12:
As far as I understood the documentation this is not possible with the current implementation of the standard template for settings dialog. So one has to create a unique template and some "knockout" code I guess. This is the tricky part - personally I did not understand this whole knockout system.

closed 2021-03-21 by stale[bot]