sethvoltz/OctoPrint-DisplayPanel #51: Progressbar is gone after starting the print

NilsRo, 2021-02-06:
Hi.

After starting the print the progressbar is gone with new version 2.0. It seems to be related to the estimation of the remaining time. If it is not calculated yet in OctoPrint Webfrontend the progressbar is not visible. As it is visible before the print starts it should stay on 0% or should be invisible also on the print start.

[image]

Regards,
Nils

unclej84, 2021-02-12:
Hello @NilsRo,

this is really strange. Is it missing on all screens? What settings do you use (Rotation, Progress on Top....)? Was it still heating up at this point or already printing? Are you using any special plugins?

NilsRo, 2021-02-12:
Hi. 

Everything is set to standard as I updated to V2 Plug-in and changed the display format of the end time only. The bar is only gone during heat up. Before and after it is visible. There is no special plug-in that changing this information but I plan to use one to better calculate the remaining time. But not important here.

unclej84, 2021-02-12:
Is there any output from the plugin in octoprint.log?

NilsRo, 2021-02-12:
Nothing in the log. But I did not enable a debuglevel or something else.

NilsRo, 2021-02-20:
Hi,

I setup a dev environment to develop my own plugin and found this during virtual printing. So it seems matching the issue on my real OctoPrint perhaps. Seems to be a tpe mismatch.

alueError: could not convert string to float: '-'
2021-02-20 14:51:25,821 - octoprint.plugins.display_panel - WARNING - An exception of type TypeError occurred on log_error. Arguments: ("argument of type 'NoneType' is not iterable",)
2021-02-20 14:51:25,821 - octoprint.plugins.display_panel - WARNING - An exception of type TypeError occurred on log_error. Arguments: ("unsupported operand type(s) for +: 'int' and 'NoneType'",)
2021-02-20 14:51:25,926 - octoprint.plugin - ERROR - Error while calling plugin display_panel
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_display_panel/__init__.py", line 159, in on_event
    self._displaylayerprogress_total_height = float(payload.get('totalHeight'))
ValueError: could not convert string to float: '-'
2021-02-20 14:51:26,149 - octoprint.plugin - ERROR - Error while calling plugin display_panel
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_display_panel/__init__.py", line 159, in on_event
    self._displaylayerprogress_total_height = float(payload.get('totalHeight'))
ValueError: could not convert string to float: '-'
2021-02-20 14:51:26,173 - octoprint.plugins.display_panel - WARNING - An exception of type TypeError occurred on log_error. Arguments: ("argument of type 'NoneType' is not iterable",)
2021-02-20 14:51:26,174 - octoprint.plugins.display_panel - WARNING - An exception of type TypeError occurred on log_error. Arguments: ("unsupported operand type(s) for +: 'int' and 'NoneType'",)
2021-02-20 14:51:26,288 - octoprint.plugin - ERROR - Error while calling plugin display_panel
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_display_panel/__init__.py", line 159, in on_event
    self._displaylayerprogress_total_height = float(payload.get('totalHeight'))
ValueError: could not convert string to float: '-'

sethvoltz, 2021-02-28:
I just pushed up v2.1.0, can you test again after you upgrade? I think I may not have included some safety code around this in the earlier releases. If it persists, I'll take another look into this.

Thanks!

sethvoltz, 2021-02-28:
Also, I dig the color and placement of your build. You should post a picture in the [Case Design Show and Tell](https://github.com/sethvoltz/OctoPrint-DisplayPanel/discussions/32)

NilsRo, 2021-03-01:
Hi. There error is still there. It happens in the case no estimation is available. The GUI looks like this and there are also Typ Conversion errors in the log. In my opinion check is missing if the estimation is "None".  

[image]
[image]

sethvoltz, 2021-04-12:
Whoops, didn't want this to get closed. Bad robot. @NilsRo been super busy at work but this is still on my radar to dig into

NilsRo, 2021-05-07:
I now get the following error continuously during print. But the bar is visible. Still updated to newest V3.

2021-05-07 01:46:10,335 - octoprint.plugins.display_panel - WARNING - An exception of type TypeError occurred on log_error. Arguments: ("argument of type 'NoneType' is not iterable",)

NilsRo, 2021-07-01:
Hi,

progressbar is there now but the exception is still there and producing much errors in the log. Do you have an idea how to solve this?

Cheers,
Nils
[octoprint(2).log](https://github.com/sethvoltz/OctoPrint-DisplayPanel/files/6750520/octoprint.2.log)

closed 2021-08-07 by stale[bot]