sethvoltz/OctoPrint-DisplayPanel #54: Installation issue

Pele79, 2021-02-10:
I am having the same or similar issue to #17 on an RPi4 4gb version on a fresh install of the latest Octopi release, I did have this issue on the previous version which I had updated to Python 3 manually, but when I couldn't get it to work I decided to try a fresh install to no avail. If anyone has a fix that would be greatly appreciated.

lgreif, 2021-02-10:
I am also having trouble getting this up and running on an OctoPi image on RPI4 (not sure on the RAM, probably a lower spec). I followed the install instructions and the "3C" is showing up in the correct spot when I run the detection command.  Any help with this would be awesome!

Pele79, 2021-02-11:
Hi all, I found the fix on the Octoprint forum [https://community.octoprint.org/t/octoprint-displaypanel-not-installing/30210/6](url) , it seems the rpi.gpio is not installing in the correct place.  I will leave the issue open so @sethvoltz can see it.

sethvoltz, 2021-02-28:
Merged @unclej84's PR. I'll cut a release in a little bit. Thanks everyone!

closed 2021-02-28 by sethvoltz