sethvoltz/OctoPrint-DisplayPanel #58: Pause on GPIO 27 gets activated without button press

Harryhar, 2021-02-26:
Pause gets activated without me pressing the button. If I set the pause to another GPIO (26), the pause function is not triggerd anymore (my button is still connected to GPIO27).

When I use the command "raspi-gpio get 27" from the commandline, the value displayed shows the correct button state. This way I checked if the wiring is correct.

Why is the plugin not responding like it should? The other buttons are working without any problems.

I first thought it was a conflict with the enclosure plugin but after disabling that plugin the problem still persists.

Any help would be appreciated.

sethvoltz, 2021-02-28:
We've seen this when there are issues with the wiring to the GPIO pin. Long wires without a hardware debounce and where the wired connection is poor, such as a loose plug that wiggles with the printer movement. Please check your wiring

timinator2020, 2021-03-09:
I had the same issue with pause using a breadboard and cheap jumper wires (but only with pause, and only on GPIO27).  

Software debouncing can only go so far.  I added a 10k pull-up resistor between 3v and each GPIO pin, and a 0.1uF capacitor between each GPIO pin and ground. That seemed to take care of the issue for me (so far).  I also added a 750 ohm current-limiting resistor inline with the GPIO pin to prevent accidentally frying the GPIO in the event that pin gets switched to be an output or something (never can be too careful).  (It's the equivalent of the 1k resistor in the schematic linked below)

A quick google search should reveal plenty of other options around false triggering and hardware debouncing.  Here's one example:
https://hackaday.com/2015/12/09/embed-with-elliot-debounce-your-noisy-buttons-part-i/
See the 3rd picture/schematic for a basic representation of what I did.

The Raspberry Pi should already have pull-up resistors enabled, so you may also get by with just adding the 0.1uF Caps to ground.

sethvoltz, 2021-03-28:
Since we have a discussion into handling these issues, let's move the conversation over there. Please post in the discussion if you continue to have odd button press issues.

[Hardware Installation Issues and Remedies](https://github.com/sethvoltz/OctoPrint-DisplayPanel/discussions/36)

closed 2021-03-28 by sethvoltz