sethvoltz/OctoPrint-DisplayPanel #89: Support for 1.3" 128X64 OLED display

osvobo, 2022-02-08:
Hello Seth,

would it be difficult to get support also for bigger, 1.3" displays, such as this one?

https://www.amazon.com/dp/B06XXTHLNW/ref=sspa_dk_detail_5?psc=1&pd_rd_i=B06XXTHLNW&pd_rd_w=YxSAI&pf_rd_p=887084a2-5c34-4113-a4f8-b7947847c308&pd_rd_wg=k8kng&pf_rd_r=QTXY4YZEMH9Y01NK8MZ2&pd_rd_r=dada1f44-781e-440c-87d6-5f3aa907706c&s=industrial&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFLN05HU0RUNUZVVkImZW5jcnlwdGVkSWQ9QTAxNTc4NjUyQ1JJUFI4TUtRN1VTJmVuY3J5cHRlZEFkSWQ9QTA2Mjg4NTdQM0hEOUhCVjk3TUQmd2lkZ2V0TmFtZT1zcF9kZXRhaWwmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

Thanks,

Ondrej

kgutwin, 2022-02-08:
It looks like that display is based off the SH1106 driver chip, and there are some discussions on that topic in discussion #86. None of that code has been submitted as a PR or change yet, so while it might not be much work to do, there isn't any official support for those displays yet.

osvobo, 2022-02-08:
Ok, thanks and sorry that I completely overlooked https://github.com/sethvoltz/OctoPrint-DisplayPanel/discussions/86. It would be nice if that display got official support in the future.

closed 2022-03-17 by stale[bot]