sethvoltz/OctoPrint-DisplayPanel #95: Plugin not working with Octoprint 1.9.3

stripu, 2023-11-04:
Just installed a fresh octopi and connected the oled and I have no output on it. 
In the logs there's a bunch of errors like:

2023-11-05 00:06:28,830 - octoprint.plugins.display_panel - WARNING - An exception of type AttributeError occurred on log_error. Arguments: ("'ImageDraw' object has no attribute 'textsize'",)

2023-11-05 00:06:48,673 - octoprint.plugin - ERROR - Error while calling plugin display_panel
Traceback (most recent call last):
  File "/home/quiztech/oprint/lib/python3.9/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/quiztech/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/quiztech/oprint/lib/python3.9/site-packages/octoprint_display_panel/__init__.py", line 88, in on_event
    self.set_printer_state(event)
  File "/home/quiztech/oprint/lib/python3.9/site-packages/octoprint_display_panel/__init__.py", line 284, in set_printer_state
    self.disp.update_timer(self._printer_state)
AttributeError: 'Display_panelPlugin' object has no attribute 'disp'

I've tried the oled screen with a different plugin - https://github.com/fredrikbaberg/OctoPrint-SSD1306 - and it seems to work, but the other plugin doesn't have as many details as I would prefer :)

Any idea what could be causing this issue?
Got some free time to update this for 1.9.3?

Thanks for a great plugin! Here's hoping you can help us again :)

feserdel, 2023-11-07:
Same error happens to me. The screen is always black.

Log shows WARNINGS like:

2023-11-06 13:23:41,046 - octoprint.plugins.display_panel - WARNING - An exception of type AttributeError occurred on log_error. Arguments: ("'ImageDraw' object has no attribute 'textsize'",)

2023-11-07 08:54:54,733 - octoprint.plugins.display_panel - WARNING - An exception of type ValueError occurred on log_error. Arguments: ('x1 must be greater than or equal to x0',)

osvobo, 2023-11-13:
Hi, my screen is also black (both, virtual as well as the physical one). I'm getting the following:

2023-11-12 23:00:50,652 - octoprint.plugins.display_panel - INFO - Initializing screens...
2023-11-12 23:00:50,794 - octoprint.plugins.display_panel - WARNING - An exception of type AttributeError occurred on log_error. Arguments: ("'ImageDraw' object has no attribute 'textsize'",)
2023-11-12 23:00:50,799 - octoprint.plugins.display_panel - WARNING - An exception of type AttributeError occurred on log_error. Arguments: ("'ImageDraw' object has no attribute 'textsize'",)

...this warning repeats itself from here

OFreddy, 2023-12-24:
Hi. 
I had the same problem which was a pitty as I am using this great plugin for years.
I found out, that outdated methods were used that are no longer supported by a library in current versions.

You can try using the changes in my fork. Works fine for me. 
Only the used default font became another with slightly larger letters. 
The font is also not monospaced anymore. 

It can be installed manually using this URL:
```none
https://github.com/OFreddy/OctoPrint-DisplayPanel/archive/main.zip
```

closed 2024-03-13 by stale[bot]